gamaral/rpi-buildroot #29: Support Raspberry Pi 2

probonopd, 2015-02-14:
It would be nice if we could have an optimized build that takes full advantage of the new CPU in the Raspberry Pi 2.

ccrisan, 2015-02-14:
That's what I have in mind. As a matter of fact I'm working right now on the new version. My plans are that it works well on RaspberryPI, RaspberryPI 2 and BananaPI.

probonopd, 2015-02-14:
Excellent, looking forward to it.

gamaral, 2015-02-15:
@probonopd is this related to rpi-buildroot? If so, then yes. I'm working on it. :)

probonopd, 2015-02-15:
Indeed I was talking about rpi-buildroot. But interesting to see this commit for motionPie, https://github.com/ccrisan/motionPie/commit/0d16283d50b104ef1ed43118f9dc5da025a8c601

andor-pierdelacabeza, 2015-03-07:
I'd love that. I have several RPi 2 I'd be happy to use for testing.

I can't code but I can test and document :laughing: :ok_hand:

gamaral, 2015-03-08:
It's on it's way. I'm almost done combining with the upstream project. :) It's just gonna take some time since I have a really bad internet connection ATM.

gamaral, 2015-03-08:
defconfig script for RPi2 tested and added test-drive image for it.

closed 2015-03-08 by gamaral

andor-pierdelacabeza, 2015-03-10:
Tested  and working! During the will I'll push the limits of it a bit and see how it behaves :D :+1: