gamaral/rpi-buildroot #33: Building midori not possible?

probonopd, 2015-03-21:
I am trying to build a fast-booting digital signage solution using rpi-buildroot.
However, I get
```
Graphic libraries and applications (graphic/text)  ---> 
*** midori needs libgtk2 and a toolchain w/ C++, wchar, threads ***
```
I see no way to select "libgtk2 and a toolchain w/ C++, wchar, threads".

gamaral, 2015-03-22:
You need to turn on C++, wchar and threads in the toolchain. Read the buildroot docs (link in README.md) ;)

closed 2015-03-22 by gamaral

probonopd, 2015-03-22:
Sorry for being slightly OT, but even after reading the buildroot docs it is unclear to me what options exactly are missing in order to fulfill "libgtk2 and a toolchain w/ C++, wchar, threads". I don't find GTK2 (only GTK3) as an option in `make nconfig`and I have tried with glibc (assuming it would have C++, wchar, threads). When this didn't work, I reverted back to uClibc and selected C++ and wchar in `make nconfig` but don't seem to have an option to select threads. 

Sorry if this is beyond the scope of rpi-buildroot, but I figured that many users of rpi-buildroot might be interested in using it for a simple digital signage solution.

gamaral, 2015-03-22:
NP, I'll give you another hint: libgtk2 requires Xorg, you will need to enable that package for the libgtk2 package to even show up.

probonopd, 2015-03-23:
Thanks for the hint.