gamaral/rpi-buildroot #34: python-flask: "ImportError: No module named zlib"

013ez, 2015-03-22:
Hi, I'm trying to run a web-server using python-flask. 
Host: Linux Mint
Target: rpi 1

I make the following steps on host:
1. cd /home/n1/test
2. git clone --depth 1 git://github.com/gamaral/rpi-buildroot.git
3. cd rpi-buildroot
4. make raspberrypi_defconfig
5. make nconfig         # add nano, python (2.*), python-flask (zlib added auto) ,wchar
6. make
7. sudo board/raspberrypi/mksdcard /dev/sdc

on rpi:
1. cd /home
2. nano test.py:
   from flask import Flask
   app = Flask(__name__)
   @app.route("/")
   def hello():
       return "Hello World!"
   app.run()
3. python test.py
Get error:
```python
   traceback (most recent call last):
      File "test.py", line 1 in, in <module>
         from flask import Flask
      File "home/n1/test/rpi-buildroot/output/target/usr/lib/python2.7/site-packeges/flask/__init__.py, line 17, in <module>
      File "home/n1/test/rpi-buildroot/output/target/usr/lib/python2.7/site-packeges/werkzeug/__init__.py, line 154, in <module>
...
   ImportError: No module named zlib
```

I think problem with python lib path. With python3 on step 6:Unknown distribution option: 'zip_safe'
Sorry for my English. Thank you.

gamaral, 2015-03-22:
Yup, looks like python is it's storing your host's installation path. That's an upstream bug though, I'll see if it's already reported.

You can work around it though right? I think you can set PYTHONPATH or some other env variable.

013ez, 2015-03-23:
I can't fix it. I don't see my hostpath in env vars.

gamaral, 2015-03-23:
It's compiled in, but you should be able to override it using env variables.

013ez, 2015-03-24:
I set PYTHONPATH, but it doesn't work. [Here the normal path.](http://git.buildroot.net/buildroot/commit/?id=82e23f15a11c2cd66d0f6c6397906663c9c9e1d7) Now I work with bottle.py

013ez, 2015-04-01:
Problem with locale settings. Exporting LANGUAGE=en_US helped. Wait official patch.

closed 2015-04-01 by 013ez