gamaral/rpi-buildroot #36: mksdcard does not work with devices that use 'p' as their partition separator

lilphil, 2015-03-22:
mksdcard does not work with devices that use 'p' as their partition separator, eg:

    sudo ./board/raspberrypi/mksdcard /dev/mmcblk0

    *****************************************************************************************
    > Formatting partitions...
    *****************************************************************************************
    mkfs.fat 3.0.26 (2014-03-07)
    /dev/mmcblk01: No such file or directory

jolyboy, 2015-09-26:
I also had this problem. I got around it by changing the following lines

${MKFS_VFAT} -F 16 -n BOOT -I "${SDCARD}1" || exit 1
${MKFS_EXT4} -F -q -L rootfs "${SDCARD}2" || exit 1
${MOUNT} "${SDCARD}1" .mnt || exit 2
${MOUNT} "${SDCARD}2" .mnt || exit 2

to 

${MKFS_VFAT} -F 16 -n BOOT -I "${SDCARD}p1" || exit 1
${MKFS_EXT4} -F -q -L rootfs "${SDCARD}p2" || exit 1
${MOUNT} "${SDCARD}p1" .mnt || exit 2
${MOUNT} "${SDCARD}p2" .mnt || exit 2

gamaral, 2015-11-21:
Should be solved now.

closed 2015-11-21 by gamaral