gamaral/rpi-buildroot #40: Test Drive Build - No Input from keyboard.

iamjoshlee, 2015-08-30:
This could just be me missing something incredibly obvious. However, I've loaded the test drive build on to my sd card and everything boots as it should. However, if I connect a usb keyboard and boot, after reaching the login prompt, it detects that there's a new usb input, etc. Yet, it doesn't register any input from the keyboard, nor does it ever return to prompt.

gamaral, 2015-08-30:
Ah yes. It's missing event input support. It was meant to be interacted
through the UART. :)

iamjoshlee, 2015-08-30:
ah got it!

kimmobrunfeldt, 2015-11-18:
I also noticed this problem. I tried to do a new manual build where I enabled bunch of configurations, such as `input-event-daemon` and `input-tools` but I couldn't figure out which configuration I actually need to enable for USB keyboard support. Do you know which ones need to be enabled or is this something out of buildroot's scope.

kimmobrunfeldt, 2015-11-19:
I got this fixed by changing my keyboard to another.

closed 2015-12-16 by gamaral