gamaral/rpi-buildroot #41: Failed to open vchiq instance. Raspistill

HLK21, 2015-09-20:
I get this error by starting raspistill.
             Failed to open vchiq instance

* Fixed by changing rpi userland and rpi firmware to latest commit.

akaer, 2015-09-21:
Same for me

gamaral, 2015-09-21:
I'll give it a try Tonight. You guys selected the _x firmware right?

HLK21, 2015-09-22:
Selected packages:

Target packages/Audio and video applications/ffmpeg
Target packages/Hardware handling/rpi-userland
Target packages/Hardware handling/Firmware/rpi-firmware/Firmware to boot/extended
Target packages/Libraries/Graphics/jpeg support
Target packages/Libraries/Hardware handling/lib4vl
Target packages/Networking applications/wireless tools
Target packages/Networking applications/wpa_supplicant
Target packages/Networking applications/wpa_supplicant/Enable EAP
Target packages/Networking applications/wpa_supplicant/Enable syslog support
Target packages/Networking applications/wpa_supplicant/Install wpa_cli binary
Target packages/Networking applications/wpa_supplicant/Install wpa_passphrase binary

akaer, 2015-09-22:
I'm using the standard Raspberry Pi B (not B+). Parts from my custom defconfig can be found here https://gist.github.com/akaer/4a96308c3a652720b08b

HLK21, 2015-09-22:
I am using raspberrypi2_defconfig

jolyboy, 2015-09-27:
I had this problem also. Included x firmware but no luck. Then I went back to 4.0.y firmware and it worked.

gamaral, 2015-09-27:
Okay, bumped rpi-firmware and rpi-userspace to master, I must have messed up a merge at some point. I tested it Today on the RPI A/B and RPI 2, worked a treat. :)

closed 2015-09-27 by gamaral