gamaral/rpi-buildroot #42: vc API error while loading shared libraries: libilclient.so

lecheel, 2015-10-02:
hi,
After select rpi-userland and hello_pi examples
what should I do for following error
[image]

Thanks 
Lechee

gamaral, 2015-12-16:
I think this was caused rpi-userland, it's been updated since. Feel free to reopen if it crops up again.

closed 2015-12-16 by gamaral

schenkmi, 2016-02-08:
I have the same issue and I guess there is a typo in rpi-userland.mk which installs libilclient.so to libilcient.so. So the linker will never find it on the target

Correct:
define RPI_USERLAND_EXTRA_LIBS_TARGET
	$(INSTALL) -m 0644 -D \
		$(@D)/build/lib/libilclient.so \
		$(TARGET_DIR)/usr/lib/libilclient.so
endef

Cheers

Michael