gamaral/rpi-buildroot #44: Should not use symbolic versions in rpi-firmware.mk / rpi-userland.mk

thomask77, 2015-10-16:
Hi!

I saw that you use

    RPI_FIRMWARE_VERSION = master
    RPI_USERLAND_VERSION = master

Instead of full commit IDs as in the original buildroot.

According to http://free-electrons.com/doc/training/buildroot/buildroot-slides.pdf, p. 105, this is bad practice:

*"always use a tag name or a full commit id, and never a branch name: the code will never be re-downloaded when the branch is updated."*

Also, have you tried to submit your other changes (i.e. mksdcard/mkimage, hardfloat) upstream?

best regards,
Thomas

gamaral, 2015-10-16:
Yup, I used to specify commit id's back before buildroot had RPi support. I stopped since using commit-id don't get shallow cloned, branches and tags do. Unfortunately the RPi folks have some commitment issues and don't want to tag anything; I was maintaining clones of their repos which I would tag and everything in order to support shallow-cloning, but I have no time to keep them up to date. Long story short, using master is the next best thing ATM.

Anyway, this overlay is suppose to be bleeding-edge, so users that do a distclean don't have to worry about the checkout not getting updated, and if you're "hard-core" you can remove the tar files from the 'dl' directory.

As for upstreaming, I did try a few times.

gamaral, 2015-12-16:
Switched to stock packages, so this is not an issue anymore.

closed 2015-12-16 by gamaral