gamaral/rpi-buildroot #49: Raspberry pi zero

Gatux, 2016-03-03:
Hi,

Thanks a lot for your work :)
I tried the images on my raspberry pi zero, however I coundn't make it work.
Did you tried or do you have any idea how it could work ?

Thanks

gamaral, 2016-03-03:
I don't have one, so I can't test it. :( Sorry.

YoungCereal, 2016-03-09:
Hi for the Raspberry Zero you only need to update the firmware files in the boot Partition.

Gatux, 2016-03-11:
Hi
I cloned the project and built it  with default config (raspberrypi_defconfig) but it would'nt work first try : keyboard works but could not do anything more.
However, using your sdimage and with what YoungCereal mentioned, I managed to make it work by changing these only two files in the boot partition : bootcode.bin and start.elf
I will continue to play with this when I have more time. Thanks.

gamaral, 2016-03-11:
If you cloned the project and built the raspberrypi_defconfig, you shouldn't need to update the boot files, they should be updated by default.

The defconfigs I supply use the UART as the default tty. You may need to nuke the cmdline.txt file in the boot partition.

I pushed an update that will ignore the cmdline.txt in the future. :)

Gatux, 2016-03-11:
Ok this may be why I couldnt do anything ^^
I'll try again this weekend.
By the way, is it possible to have the config you used for your images ? (ex: http://dl.guillermoamaral.com/rpi/sdcard.img.xz)

Gatux, 2016-03-12:
Tried again today, I still had to change “TTY port” from "ttyAMA0" to "tty1" in order to have direct shell, but it works :) Thanks ^^
I can now play with it and create the perfect image for my needs :)

gamaral, 2016-03-12:
That's what I meant by it being configured to use UART as the default console. Glad you're sorted. :)

closed 2016-03-12 by gamaral