gamaral/rpi-buildroot #51: output/images/rootfs.tar not found

bphermansson, 2016-03-31:
Tried to use the mksdcard script but it fails: "Didn't find rpi-firmware and/or rootfs.tar! ABORT."
Then tried the manual instructions which also fails:

sudo tar -xvpsf output/images/rootfs.tar -C /media/rootfs
tar: output/images/rootfs.tar: Funktion "open" misslyckades: Filen eller katalogen finns inte
tar: Error is not recoverable: exiting now

And for sure there's no file "rootfs.tar". What's wrong?

qnl3, 2016-03-31:
You have to configure you build to output a tar image. 
using "make nconfig"
select "Filesystem Images"
check "tar the root filesystem"
rebuild.
This should produce the tar file the script uses which is the file used in the manual instructions.

bphermansson, 2016-03-31:
Ok thanks, that works. Now I just have to find out how to get a command prompt at the Pi.

gamaral, 2016-04-01:
Switch the main console from the UART to TTY :)

gamaral, 2016-04-01:
Also, if you checked out the latest tag, an image should be created for you automatically. :)

bphermansson, 2016-04-01:
Nice! I set the TTY to "tty1" and got a login prompt. All good.

closed 2016-04-08 by bphermansson