gamaral/rpi-buildroot #53: WiFi adapter detected but not being used

ilendemli, 2016-04-22:
i couldn't find anything related to it so i am opening this issue
searching the internet, i found two similar tutorials getting wifi to work using buildroot.
the tutorials are a bit old so some screens changed, but comparing them showed that the default settings are actually already set and it should work outside the box

the raspberry pi recognized the dongle, and it (vid = 7392, pid = 7811) shows up when i do lusb, also dmesg shows the dongle properly

even after editing /etc/network/interfaces and adding a wlan0 interface, if won't show up when i do ifconfig -a

do you have any idea how i could fix this?

gamaral, 2016-04-22:
1) Make sure your kernel is building the module.
2) Load the module (buildroot won't autoload the module for you).

Alternatively, compile the module into the kernel.

ilendemli, 2016-04-22:
1) i thinks it's building it but i am not sure, gonna check it
2) i will try that if 1) does actually work

thank you for answering that fast

ilendemli, 2016-04-22:
Alright, detected WiFi adapter. the module was build but not loaded
"modprobe 8192cu" did the job
thanks for your help

closed 2016-04-24 by ilendemli