gamaral/rpi-buildroot #54: Investigate using an overlay without pulling the whole Buildroot source tree in

voltagex, 2016-05-01:
See https://github.com/baxterworks-build/buildroot-skeleton - just an idea. Will make upgrading buildroot easier in future.

gamaral, 2016-05-03:
Interesting idea, know if it supports packages and not just configs?

voltagex, 2016-05-05:
Yep, use $BR2_EXTERNAL. Happy to help with this if you can show me where your customisations have been made.

mbrgm, 2016-08-20:
@voltagex As I'd also be interested in having an up-to-date version of buildroot, I started by diffing the last merged version of buildroot with the current state of this repo:

This is the result I get from `git diff aaf6c28..HEAD`:

```patch
diff --git a/README.md b/README.md
new file mode 100644
index 0000000..25d3ee3
--- /dev/null
+++ b/README.md
@@ -0,0 +1,121 @@
+Buildroot for Raspberry Pi
+==========================
+
+This buildroot *overlay* will produce a bleeding-edge, light-weight and trimmed
+down toolchain, rootfs and kernel for the Raspberry Pi. It's intended for
+**advanced users** and specific embedded applications.
+
+Before You Begin
+----------------
+
+- If you're not familiar with Buildroot and what it can and can't do, please
+  take the time to [read the manual](http://buildroot.org/downloads/manual/manual.html).
+
+- You must be pretty comfortable with **cross-compilation** in order to use
+  rpi-buildroot.
+
+Test Drive
+----------
+
+You can test drive rpi-buildroot by following the instructions below:
+
+	wget http://dl.guillermoamaral.com/rpi/sdcard.img.xz
+	xz -d sdcard.img.xz
+	sudo dd if=sdcard.img of=/dev/sdx # replace *sdx* with your actual sdcard device node
+
+There's now also a test drive image for the Raspberry Pi 2:
+
+	wget http://dl.guillermoamaral.com/rpi/sdcard2.img.xz
+	xz -d sdcard2.img.xz
+	sudo dd if=sdcard2.img of=/dev/sdx # replace *sdx* with your actual sdcard device node
+
+The default user is **root**, no password is required.
+
+Building
+--------
+
+	git clone --depth 1 git://github.com/gamaral/rpi-buildroot.git
+	cd rpi-buildroot
+	make raspberrypi_defconfig # if your target is a Raspberry Pi 2, use 'raspberrypi2_defconfig'
+	make nconfig         # if you want to add packages or fiddle around with it
+	make                 # build (NOTICE: Don't use the **-j** switch, it's set to auto-detect)
+
+Deploying
+---------
+
+### Script
+
+I've added a script that can automatically flash your sdcard, you simply need
+to point it to the correct device node, confirm and you're done!
+
+**Notice** you will need to replace *sdx* in the following commands with the
+actual device node for your sdcard.
+
+    # run the following as root (sudo)
+    board/raspberrypi/mksdcard /dev/sdx
+
+### Manual
+
+You will need to create two partitions in your sdcard, the first (boot) needs
[... 535 lines cut ...]
 BR2_LINUX_KERNEL_ZIMAGE=y
 
 # Build the DTBs for A/B, A+/B+ and compute module from the kernel sources
diff --git a/package/rpi-firmware/rpi-firmware.mk b/package/rpi-firmware/rpi-firmware.mk
index 4b92a50..4846297 100644
--- a/package/rpi-firmware/rpi-firmware.mk
+++ b/package/rpi-firmware/rpi-firmware.mk
@@ -4,7 +4,7 @@
 #
 ################################################################################
 
-RPI_FIRMWARE_VERSION = 22d8c910f4c53118f9cf85c038d7d8e307efc110
+RPI_FIRMWARE_VERSION = 1.20160309
 RPI_FIRMWARE_SITE = $(call github,raspberrypi,firmware,$(RPI_FIRMWARE_VERSION))
 RPI_FIRMWARE_LICENSE = BSD-3c
 RPI_FIRMWARE_LICENSE_FILES = boot/LICENCE.broadcom
@@ -41,7 +41,6 @@ define RPI_FIRMWARE_INSTALL_IMAGES_CMDS
 	$(INSTALL) -D -m 0644 $(@D)/boot/start$(BR2_PACKAGE_RPI_FIRMWARE_BOOT).elf $(BINARIES_DIR)/rpi-firmware/start.elf
 	$(INSTALL) -D -m 0644 $(@D)/boot/fixup$(BR2_PACKAGE_RPI_FIRMWARE_BOOT).dat $(BINARIES_DIR)/rpi-firmware/fixup.dat
 	$(INSTALL) -D -m 0644 package/rpi-firmware/config.txt $(BINARIES_DIR)/rpi-firmware/config.txt
-	$(INSTALL) -D -m 0644 package/rpi-firmware/cmdline.txt $(BINARIES_DIR)/rpi-firmware/cmdline.txt
 	$(RPI_FIRMWARE_INSTALL_DTB)
 	$(RPI_FIRMWARE_INSTALL_DTB_OVERLAYS)
 endef
```

MrAda, 2016-08-20:
I think gamaral is on vacation.  I hope when he gets back we can get some help with this.  I have yet to get a build root creation to work on my RPi3

gamaral, 2016-08-20:
RPis are pretty well supported in vanilla Buildroot now. Images get created and everything. ;-) I don't really think you guys need to go through the trouble. ;-)

MrAda, 2016-08-20:
That isn't true.  It only supports HDMI output, I don't have a device to output on unless maybe my TV.  I want to get it to boot fast and then work on the driver for the LCD panel I have, for which I have no technical information.

gamaral, 2016-08-20:
If you need TTY to be on the UART, you can change that in the standard config.

I.e. BR2_TARGET_GENERIC_GETTY_PORT="ttyAMA0"

voltagex, 2016-08-20:
@mbrgm can you trim that post and use attachments or gists for that amount of text? It's making the discussion hard to follow

MrAda, 2016-08-20:
@gamaral I don't understand.  What does TTY or UART have to do with the LCD screen?  That sounds like a keyboard kind of interface.  The raspberry pi official 7" screen with touch is connect to the Pi by an DSI ribbon cable.

gamaral, 2016-08-21:
Maybe I misunderstood "That isn't true. It only supports HDMI output". Did you mean it doesn't support RCA then?

MrAda, 2016-08-21:
No i didn't mean RCA, didn't realize there was an RCA out.  What I have is the official Raspberry pi LCD display and it has an DSI ribbon connector.

voltagex, 2016-08-22:
Hasn't this gone a bit off topic now?

gamaral, 2016-08-22:
I have to agree. In any case, like I said, most changes in this report are no longer required to get a system up and working. I would say stick with Buildroot proper instead. ;-)

gamaral, 2016-08-22:
Sorry, not report, repo. Autocorrect. :-(

voltagex, 2016-09-07:
Closing this as I'm no longer using this code.

closed 2016-09-07 by voltagex