gamaral/rpi-buildroot #57: Raspberry Pi 3 support

MrAda, 2016-07-23:
I'd like to be able to use what you have done to use on my Pi 3.

gamaral, 2016-11-02:
Yup, you're right. It ain't budging. Even with Raspbian's kernel, I can't get anything in the UART. My guess is that there's a dtb config issue. I'll need to play around with it some more this weekend.

MrAda, 2016-11-02:
Your efforts are appreciated!

gamaral, 2016-11-02:
Confirming it boots fine. Just no UART output.

gamaral, 2016-11-02:
Okay, apparently they are using PL011 for BT, you would need to tweak the config.txt to get it working. Enable mini UART and fix the clock so that you don't get corruption. :-/

https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=141195

MrAda, 2016-11-02:
Do you have you code and changes in a GIT repository?  I couldn't see anything working when I used buildroot.  No HDMI/Nothing.

gamaral, 2016-11-02:
Some, I'll push an update soon

GeoffreyFrogeye, 2017-03-19:
@gamaral Any update on this? Even if there are some quircks, is it possible for you to push your work in a separate branch? Thanks.

MrAda, 2017-03-19:
@GeoffreyFrogeye I'd be interested still myself.  I have been able to get a buildroot version working but it is still slow as compared to @gamaral 's version.

gamaral, 2017-03-19:
IIRC there was a fix pushed up for this in the RPi kernel repository, which you would get in a clean build.
I updated the repository, so it should contain it now (untested ATM, feel free to test it out though)..

MrAda, 2017-03-20:
Ok.  I have tried to build this today and it seem that the make finished fine.  I was unable to use the script to load the stuff onto the sdcard and then when I did it manually I got stuck at the point were I am suppose to untar the rootfs.tar file. The make never build the rootfs.tar.  I am stuck.

gamaral, 2017-03-20:
Use the img file instead if you're having trouble with the mksdcard script. ;-)
(I updated the README).

MrAda, 2017-03-20:
@gamaral I feel that I was successful after both your suggestions and @GeoffreyFrogeye.

The boot time, after the display did it's 2-3 second splash, took about 7 seconds to get to a login prompt.  Granted I was  hoping for that 1-2 second boot up time like you did with the RPi2.  Any idea why it takes that long?

gamaral, 2017-03-27:
I trimmed the hell out of the kernel and the OS, I also made the image boot straight into my game engine and kept loading non essential bits in the background. :-)

isage, 2017-11-10:
@gamaral can you provide your configs?

janemxq, 2017-12-13:
i buy a pi 3 develop mainboard from www.taobao.com,then can i use this bulidroot to create system image,then install into my mainboard?

gamaral, 2018-01-20:
Supported

closed 2018-01-20 by gamaral