gamaral/rpi-buildroot #60: Unable to build on Arch Linux

MrHaroldA, 2016-08-15:
I'm trying run buildroot (git checkout) on an up-to-date Arch Linux, and I get this error after a while:

```
In file included from ../../gcc/cp/except.c:1013:0:
cfns.gperf: In function ‘const char* libc_name_p(const char*, unsigned int)’:
cfns.gperf:101:1: error: ‘const char* libc_name_p(const char*, unsigned int)’ redeclared inline with ‘gnu_inline’ attribute
```

After searching the Internet, it seems that this has something to do with the GCC version installed in Arch Linux:

```
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/6.1.1/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++ --enable-shared --enable-threads=posix --enable-libmpx --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --disable-multilib --disable-werror --enable-checking=release
Thread model: posix
gcc version 6.1.1 20160802 (GCC) 
```

Buildroot should be able to build on GCC 6.x just fine, according to the same internet mentioned above ;)

gamaral, 2016-08-20:
See if you can install GCC v4.9, that one seems to work fine. :-) Buildroot vanilla doesn't support 6 ATM.

MrHaroldA, 2016-08-22:
Arch only supports back to GCC 5, which also fails at the same position. I'll try to compile GCC 4 soon.

niko, 2017-01-05:
Just built it on Arch with gcc 6.2.1 and it just worked. I guess the issue can be closed. :)

gamaral, 2017-01-06:
Good to know. ;-) Thanks!

closed 2017-01-06 by gamaral