gamaral/rpi-buildroot #61: Boot Time

reysintrono, 2016-09-07:
When I test sdcard2.img.xz in my raspberry pi, it's boot in 2 or 3 seconds, but when i build a new image using rpi-buildroot, the time boot is 7or 8 seconds. 

What is the difference between Test Drive and Building?

Thank you.

MrAda, 2016-09-08:
Do you have a RPi2 or RPi3?  I suspect he customized a lot to get the times down.  I have not been successful using the sdcard2.img.xz in my RPi3.  Probably because I am using the official touchscreen/LCD.  I don't have an HDMI compatible monitor.

gamaral, 2016-09-08:
@reysintrono If you changed nothing in the configuration. You'll need to trim down the kernel and busybox.

MrAda, 2016-09-08:
I am going to try and build rpi myself.  I see @gamaral responded.  What items did you pare down?  Can you offer some assistance on what to do?

gamaral, 2016-09-08:
@MrAda Yeah, it can just be that it's for the RPi2. I also don't think I've updated it since the display support was added.

gamaral, 2016-09-08:
Pretty much removed anything that's not required by the system. Few modules, drivers that are required by the RPi build into the kernel and everything else to off.

MrAda, 2016-09-08:
@gamaral I was thinking of trying to connect to my TV.  I think I can display my Pi's output to the TV.  I'll just have to disconnect the LCD cable for the time being.  If I knew where to find the driver for that device I'd work to get it added to build root.  So trimming down the kernel and busybox is a straight forward attempt (looking at readmes and such)?  Or will it require a surgeon? :-)

gamaral, 2016-09-08:
Anyway, you'll need to play with it. :-) The default build should be pretty fast 7 secs for an RPi1 from cold start for example. That said, Unfortunately I really can't offer much support since I really have very little free time. :disappointed:

MrAda, 2016-09-08:
Ok.   I will do my best.

gamaral, 2016-09-08:
If it's the official display, I take it you only need to update the kernel and the boot files.

MrAda, 2016-09-08:
That is correct, I do have the official 7" display

gamaral, 2018-01-20:
Issue based on older repo

closed 2018-01-20 by gamaral