gamaral/rpi-buildroot #62: could not setup rpi-firmware/cmdline.txt

ruisebastiao, 2016-10-17:
Hello,

while compiling for raspberrypi 3 i'm getting this error in post-image.sh

```sh
file(rpi-firmware/cmdline.txt): stat(/home/automacao/embeddeddynamicflow/rpi-buildroot/output/images/rpi-firmware/cmdline.txt) failed: No such file or directory
vfat(boot.vfat): could not setup rpi-firmware/cmdline.txt
Makefile:680: recipe for target 'target-post-image' failed
make[1]: *** [target-post-image] Error 1
Makefile:41: recipe for target '_all' failed
make: *** [_all] Error 2
```

in fact cmdline.txt doesn't  exist and checking the package rpi-firmware the cmdline.txt file isn't installed

```sh
define RPI_FIRMWARE_INSTALL_IMAGES_CMDS
	$(INSTALL) -D -m 0644 $(@D)/boot/bootcode.bin $(BINARIES_DIR)/rpi-firmware/bootcode.bin
	$(INSTALL) -D -m 0644 $(@D)/boot/start$(BR2_PACKAGE_RPI_FIRMWARE_BOOT).elf $(BINARIES_DIR)/rpi-firmware/start.elf
	$(INSTALL) -D -m 0644 $(@D)/boot/fixup$(BR2_PACKAGE_RPI_FIRMWARE_BOOT).dat $(BINARIES_DIR)/rpi-firmware/fixup.dat
	$(INSTALL) -D -m 0644 package/rpi-firmware/config.txt $(BINARIES_DIR)/rpi-firmware/config.txt
	$(RPI_FIRMWARE_INSTALL_DTB)
	$(RPI_FIRMWARE_INSTALL_DTB_OVERLAYS)
endef
```

gamaral, 2016-10-19:
I don't have an RPi 3 so I never really modified that config file. You can try vanilla buildroot for it, it's pretty well supported.

closed 2016-10-19 by gamaral

MrAda, 2016-10-20:
In all fairness the buildroot community is not very helpful.  I have asked questions for help and no repsonse.  I don't believe that an RPi 3 version actually works.  I get about 40-50 emails a day from the GIT activities and frankly I don't think it is in a good shape to use presently.

gamaral, 2016-10-20:
Yeah, unfortunately there's pretty much no user support in the community per say.
I don't see why it wouldn't work on vanilla Buildroot, but I guess I can give it a shot. I'll need to get one though, that may take a while to get here.

gamaral, 2016-10-20:
Okay, so Amazon has it with Prime. So two days or so. Not too bad.