gamaral/rpi-buildroot #63: unterminated quoted string

pashashocky, 2016-11-05:
Hi Guillermo, 

I am having some trouble successfully running the `pong` example on my raspberry pi 2 model B with buildroot.

Firstly I cloned the `rpi-buildroot` reporistory and ran:

`make raspberrypi2_defconfig`
`make nconfig`
Enabled `X.org` and `mesa3D` and some other libraries
`make`

Everything compiled fine and I was able to boot into it on my RPI:
then ran `startx` and twm successfully appeared with some xterms spawned. 

On my main machine I then successfully compiled the `pong` example so I could run it locally following the steps from the repo for `marshmallow_h`.

After I copied the `buildroot-env` file from `marshmallow_h/resources` to `rpi-buildroot/output` tried to run it, but it gave me the following error:

```
dirname: missing operand
Try 'dirname --help' for more information.
readlink: missing operand
Try 'readlink --help' for more information.
```

After which I had to manually set the `$BUILDROOT_BASE_DIR` to the `/path/to/rpi-buildroot/output`, and the rest of the exports followed nicely.

Then back in `marshmallow_h/build` I ran:
`cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/Toolchain-buildroot.cmake -C ../cmake/Cache-raspberrypi.cmake ..`
`make clean all`

This compiled fine, but when I tried running the compiled pong binary on the RPI I got:

`./pong: line 1: syntax error: unterminated quoted string`

Please could you suggest what might be wrong here? 

I also looked at issue #27 and was able to successfully compile a hello_world.c and have it run on the RPI.

Thank you very much for your help!

pashashocky, 2016-11-06:
I have realised that i didnt actually clean the build directory of marshmallow_h before running:
`cmake -CMAKE_TOOLCHAIN_FILE=........`

The problem i am running into now is:

```
loading initial cache file ../cmake/Cache-raspberrypi.cmake
-- The C compiler identification is GNU 4.9.4
-- The CXX compiler identification is GNU 4.9.4
-- Check for working C compiler: /home/pash/pgm/rasp/rpi-buildroot/output/host/usr/bin/arm-linux-gcc
-- Check for working C compiler: /home/pash/pgm/rasp/rpi-buildroot/output/host/usr/bin/arm-linux-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/pash/pgm/rasp/rpi-buildroot/output/host/usr/bin/arm-linux-g++
-- Check for working CXX compiler: /home/pash/pgm/rasp/rpi-buildroot/output/host/usr/bin/arm-linux-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found M: /home/pash/pgm/rasp/rpi-buildroot/output/staging/usr/lib/libm.so  
-- TinyXML2 library:
	marshmallow_tinyxml2
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of off64_t
-- Check size of off64_t - done
-- Looking for fseeko
-- Looking for fseeko - found
-- Looking for unistd.h
-- Looking for unistd.h - found
-- zlib compression library:
	marshmallow_zlib
-- libpng library:
	marshmallow_libpng
-- Looking for inttypes.h
-- Looking for inttypes.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Ogg container format library:
	marshmallow_libogg
-- Vorbis audio compression library:
	marshmallow_libvorbis
	marshmallow_libvorbisfile
-- Box2D physics engine:
	marshmallow_box2d
-- Looking for clock_gettime in rt
-- Looking for clock_gettime in rt - found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Could NOT find EGL (missing:  EGL_LIBRARY) 
-- Could NOT find GLES2 (missing:  GLES2_LIBRARY) 
CMake Error at src/graphics/opengl/CMakeLists.txt:74 (message):
  OpenGL was not found and *is* very much a requirement for the subsystem
  selected.

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
EGL_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /home/pash/pgm/rasp/marshmallow_h/src/graphics/opengl
   used as include directory in directory /home/pash/pgm/rasp/marshmallow_h/src/graphics/opengl
   used as include directory in directory /home/pash/pgm/rasp/marshmallow_h/src/graphics/opengl
   used as include directory in directory /home/pash/pgm/rasp/marshmallow_h/src/graphics/opengl
   used as include directory in directory /home/pash/pgm/rasp/marshmallow_h/src/graphics/opengl
   used as include directory in directory /home/pash/pgm/rasp/marshmallow_h/src/graphics/opengl
GLES2_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /home/pash/pgm/rasp/marshmallow_h/src/graphics/opengl
   used as include directory in directory /home/pash/pgm/rasp/marshmallow_h/src/graphics/opengl
   used as include directory in directory /home/pash/pgm/rasp/marshmallow_h/src/graphics/opengl
   used as include directory in directory /home/pash/pgm/rasp/marshmallow_h/src/graphics/opengl
   used as include directory in directory /home/pash/pgm/rasp/marshmallow_h/src/graphics/opengl
   used as include directory in directory /home/pash/pgm/rasp/marshmallow_h/src/graphics/opengl

-- Configuring incomplete, errors occurred!
See also "/home/pash/pgm/rasp/marshmallow_h/build/CMakeFiles/CMakeOutput.log".
See also "/home/pash/pgm/rasp/marshmallow_h/build/CMakeFiles/CMakeError.log".
```

pashashocky, 2016-11-06:
Found that i need to enable EGL and GLES2 inside mesa3d now its 
`-- Could NOT find BCM_HOST (missing:  BCM_HOST_INCLUDE_DIR BCM_HOST_LIBRARY BCM_HOST_PTHREADS_INCLUDE_DIR BCM_HOST_khrn_client_LIBRARY BCM_HOST_vcfiled_check_LIBRARY BCM_HOST_vchiq_arm_LIBRARY BCM_HOST_vchostif_LIBRARY BCM_HOST_vcos_LIBRARY) ``

pashashocky, 2016-11-06:
Alright, i figured i needed to include rpi-userspace instead of mesa3d, now when i run it it gives me:

`pong: can't load library 'libmarshmallow_game.so`

pashashocky, 2016-11-06:
I got it to run through the demo instructions: 
```
cmake -DCMAKE_INSTALL_PREFIX=marshmallow_demos -DCOMPONENT=runtime    -P cmake_install.cmake
cmake -DCMAKE_INSTALL_PREFIX=marshmallow_demos -DCOMPONENT=deployment -P cmake_install.cmake
cmake -DCMAKE_INSTALL_PREFIX=marshmallow_demos -DCOMPONENT=demos      -P cmake_install.cmake
```

although when i start it, my screen just turns black and thats it :S

pashashocky, 2016-11-06:
I had to get rid of the requirement to check the path in the demo mian.cpp, and after that it ran fine. No input from the keyboard though unfortunately

gamaral, 2016-11-06:
I'm glad you figured out how to do it on your own. :-) I'm not really able to guide you much ATM.
As for the input, make sure you have a backend enabled (like Linux input events) and double check permissions.

pashashocky, 2016-11-06:
Yeah its a pretty nice feeling once you get it to work :+1: Thank you for the help that you are providing!

Will check Linux input events, are you running BusyBox for init or systemd?

pashashocky, 2016-11-06:
One last thing, did you get sound working w Buildroot? I tried for the whole day with alsa, but failed. Sound is a bit more important for me than input :S

gamaral, 2016-11-06:
Make sure you have Alsa enabled in the kernel and libs in buildroot. Then enable it in config.txt: http://elinux.org/R-Pi_Troubleshooting#Sound

pashashocky, 2016-11-06:
@gamaral I just got sound working, literally 2 mins ago! With a help of a guy from IRC. 

I had everything apart from that last step with `boot/config.txt` ` dtparam=audio=on` now its working! thanks for the tip! 

Lastly just the input to figure out!

pashashocky, 2016-11-07:
Ok so for those who may struggle after me :) To get the keyboard working, check that 
`make linux-nconfig`

has `Device Drivers->Input Device support -> Event Interface` set to M

then from your `rpi` you can do `modprobe evdev` and that will get the keyboard input working correctly!

@gamaral thanks for your help, will get back to you with more nuisances.

gamaral, 2016-11-07:
Hehehe no problem, I'm glad you sorted it out. That's how you learn after all. 🍜

You can also build it into the kernel and do without modprobe.

closed 2018-01-20 by gamaral