gamaral/rpi-buildroot #71: Cannot launch MongoDB 3.3.4

msoula, 2017-07-14:
MongoDB fails to start on a raspberry3 configuration. Here is the log when launching the application:

```
$ ./mongod 
1970-01-01T00:07:02.222+0000 F -        [main] Invalid access at address: 0
1970-01-01T00:07:02.240+0000 F -        [main] Got signal: 11 (Segmentation fault).

 0xd9b82c 0xd9a72c 0xd9b21c 0x76af1470 0xd766f0 0xd7dcb8 0xd7dc60 0xd845d4 0xd8aa00 0xd8e9cc 0x3322a8 0x33d6f0 0x33daf8 0x33dd78 0x2fe0fc 0x76adb620
----- BEGIN BACKTRACE -----
{"backtrace":[{"b":"10000","o":"D8B82C","s":"_ZN5mongo15printStackTraceERSo"},{"b":"10000","o":"D8A72C"},{"b":"10000","o":"D8B21C"},{"b":"76AC5000","o":"2C470","s":"__default_rt_sa_restorer"},{"b":"10000","o":"D666F0","s":"_ZNSt8_Rb_treeINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt4pairIKS5_N5mongo17optionenvironment5ValueEESt10_Select1stISB_ESt4lessIS5_ESaISB_EE22_M_emplace_hint_uniqueIJRKSt21piecewise_construct_tSt5tupleIJRS7_EESM_IJEEEEESt17_Rb_tree_iteratorISB_ESt23_Rb_tree_const_iteratorISB_EDpOT_"},{"b":"10000","o":"D6DCB8","s":"_ZNK5mongo17optionenvironment13OptionSection11getDefaultsEPSt3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS0_5ValueESt4lessIS8_ESaISt4pairIKS8_S9_EEE"},{"b":"10000","o":"D6DC60","s":"_ZNK5mongo17optionenvironment13OptionSection11getDefaultsEPSt3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS0_5ValueESt4lessIS8_ESaISt4pairIKS8_S9_EEE"},{"b":"10000","o":"D745D4","s":"_ZN5mongo17optionenvironment13OptionsParser16addDefaultValuesERKNS0_13OptionSectionEPNS0_11EnvironmentE"},{"b":"10000","o":"D7AA00","s":"_ZN5mongo17optionenvironment13OptionsParser3runERKNS0_13OptionSectionERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaISB_EERKSt3mapISB_SB_St4lessISB_ESaISt4pairIKSB_SB_EEEPNS0_11EnvironmentE"},{"b":"10000","o":"D7E9CC","s":"_ZN5mongo17optionenvironment46_mongoInitializerFunction_StartupOptions_ParseEPNS_18InitializerContextE"},{"b":"10000","o":"3222A8","s":"_ZNSt17_Function_handlerIFN5mongo6StatusEPNS0_18InitializerContextEEPS4_E9_M_invokeERKSt9_Any_dataOS3_"},{"b":"10000","o":"32D6F0","s":"_ZNK5mongo11Initializer7executeERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EERKSt3mapIS7_S7_St4lessIS7_ESaISt4pairIKS7_S7_EEE"},{"b":"10000","o":"32DAF8","s":"_ZN5mongo21runGlobalInitializersERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS6_EERKSt3mapIS6_S6_St4lessIS6_ESaISt4pairIKS6_S6_EEE"},{"b":"10000","o":"32DD78","s":"_ZN5mongo21runGlobalInitializersEiPKPKcS3_"},{"b":"10000","o":"2EE0FC","s":"main"},{"b":"76AC5000","o":"16620","s":"__libc_start_main"}]}
 mongod(_ZN5mongo15printStackTraceERSo+0x28) [0xd9b82c]
 mongod(+0xD8A72C) [0xd9a72c]
 mongod(+0xD8B21C) [0xd9b21c]
 libc.so.6(__default_rt_sa_restorer+0x0) [0x76af1470]
 mongod(_ZNSt8_Rb_treeINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt4pairIKS5_N5mongo17optionenvironment5ValueEESt10_Select1stISB_ESt4lessIS5_ESaISB_EE22_M_emplace_hint_uniqueIJRKSt21piecewise_construct_tSt5tupleIJRS7_EESM_IJEEEEESt17_Rb_tree_iteratorISB_ESt23_Rb_tree_const_iteratorISB_EDpOT_+0x38) [0xd766f0]
 mongod(_ZNK5mongo17optionenvironment13OptionSection11getDefaultsEPSt3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS0_5ValueESt4lessIS8_ESaISt4pairIKS8_S9_EEE+0x27C) [0xd7dcb8]
 mongod(_ZNK5mongo17optionenvironment13OptionSection11getDefaultsEPSt3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS0_5ValueESt4lessIS8_ESaISt4pairIKS8_S9_EEE+0x224) [0xd7dc60]
 mongod(_ZN5mongo17optionenvironment13OptionsParser16addDefaultValuesERKNS0_13OptionSectionEPNS0_11EnvironmentE+0x48) [0xd845d4]
 mongod(_ZN5mongo17optionenvironment13OptionsParser3runERKNS0_13OptionSectionERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaISB_EERKSt3mapISB_SB_St4lessISB_ESaISt4pairIKSB_SB_EEEPNS0_11EnvironmentE+0x110C) [0xd8aa00]
 mongod(_ZN5mongo17optionenvironment46_mongoInitializerFunction_StartupOptions_ParseEPNS_18InitializerContextE+0x58) [0xd8e9cc]
 mongod(_ZNSt17_Function_handlerIFN5mongo6StatusEPNS0_18InitializerContextEEPS4_E9_M_invokeERKSt9_Any_dataOS3_+0x18) [0x3322a8]
 mongod(_ZNK5mongo11Initializer7executeERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EERKSt3mapIS7_S7_St4lessIS7_ESaISt4pairIKS7_S7_EEE+0x184) [0x33d6f0]
 mongod(_ZN5mongo21runGlobalInitializersERKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS6_EERKSt3mapIS6_S6_St4lessIS6_ESaISt4pairIKS6_S6_EEE+0x2C) [0x33daf8]
 mongod(_ZN5mongo21runGlobalInitializersEiPKPKcS3_+0x278) [0x33dd78]
 mongod(main+0xB4) [0x2fe0fc]
 libc.so.6(__libc_start_main+0x110) [0x76adb620]
-----  END BACKTRACE  -----
1970-01-01T00:07:02.241+0000 F -        [main] /proc/self/maps:
00010000-01145000 r-xp 00000000 00:01 8065       /root/mongod
1970-01-01T00:07:02.241+0000 F -        [main] 01154000-0117e000 rw-p 01134000 00:01 8065       /root/mongod
1970-01-01T00:07:02.241+0000 F -        [main] 0117e000-011ad000 rw-p 00000000 00:00 0 
1970-01-01T00:07:02.241+0000 F -        [main] 024d8000-02546000 rw-p 00000000 00:00 0          [heap]
1970-01-01T00:07:02.241+0000 F -        [main] 768c1000-76ac5000 rw-p 00000000 00:00 0 
1970-01-01T00:07:02.241+0000 F -        [main] 76ac5000-76bea000 r-xp 00000000 00:01 5484       /lib/libc-2.24.so
1970-01-01T00:07:02.241+0000 F -        [main] 76bea000-76bfa000 ---p 00125000 00:01 5484       /lib/libc-2.24.so
1970-01-01T00:07:02.241+0000 F -        [main] 76bfa000-76bfc000 r--p 00125000 00:01 5484       /lib/libc-2.24.so
1970-01-01T00:07:02.241+0000 F -        [main] 76bfc000-76bfd000 rw-p 00127000 00:01 5484       /lib/libc-2.24.so
1970-01-01T00:07:02.241+0000 F -        [main] 76bfd000-76c00000 rw-p 00000000 00:00 0 
1970-01-01T00:07:02.241+0000 F -        [main] 76c00000-76c15000 r-xp 00000000 00:01 5432       /lib/libpthread-2.24.so
1970-01-01T00:07:02.242+0000 F -        [main] 76c15000-76c24000 ---p 00015000 00:01 5432       /lib/libpthread-2.24.so
1970-01-01T00:07:02.242+0000 F -        [main] 76c24000-76c25000 r--p 00014000 00:01 5432       /lib/libpthread-2.24.so
1970-01-01T00:07:02.242+0000 F -        [main] 76c25000-76c26000 rw-p 00015000 00:01 5432       /lib/libpthread-2.24.so
1970-01-01T00:07:02.242+0000 F -        [main] 76c26000-76c28000 rw-p 00000000 00:00 0 
1970-01-01T00:07:02.242+0000 F -        [main] 76c28000-76c44000 r-xp 00000000 00:01 5436       /lib/libgcc_s.so.1
1970-01-01T00:07:02.242+0000 F -        [main] 76c44000-76c54000 ---p 0001c000 00:01 5436       /lib/libgcc_s.so.1
1970-01-01T00:07:02.242+0000 F -        [main] 76c54000-76c55000 rw-p 0001c000 00:01 5436       /lib/libgcc_s.so.1
1970-01-01T00:07:02.242+0000 F -        [main] 76c55000-76cc2000 r-xp 00000000 00:01 5442       /lib/libm-2.24.so
1970-01-01T00:07:02.242+0000 F -        [main] 76cc2000-76cd1000 ---p 0006d000 00:01 5442       /lib/libm-2.24.so
1970-01-01T00:07:02.242+0000 F -        [main] 76cd1000-76cd2000 r--p 0006c000 00:01 5442       /lib/libm-2.24.so
1970-01-01T00:07:02.242+0000 F -        [main] 76cd2000-76cd3000 rw-p 0006d000 00:01 5442       /lib/libm-2.24.so
1970-01-01T00:07:02.242+0000 F -        [main] 76cd3000-76dbc000 r-xp 00000000 00:01 7562       /usr/lib/libstdc++.so.6.0.21
1970-01-01T00:07:02.242+0000 F -        [main] 76dbc000-76dcb000 ---p 000e9000 00:01 7562       /usr/lib/libstdc++.so.6.0.21
1970-01-01T00:07:02.242+0000 F -        [main] 76dcb000-76dd0000 r--p 000e8000 00:01 7562       /usr/lib/libstdc++.so.6.0.21
1970-01-01T00:07:02.242+0000 F -        [main] 76dd0000-76dd3000 rw-p 000ed000 00:01 7562       /usr/lib/libstdc++.so.6.0.21
1970-01-01T00:07:02.242+0000 F -        [main] 76dd3000-76dd5000 rw-p 00000000 00:00 0 
1970-01-01T00:07:02.242+0000 F -        [main] 76dd5000-76dd7000 r-xp 00000000 00:01 5490       /lib/libdl-2.24.so
1970-01-01T00:07:02.242+0000 F -        [main] 76dd7000-76de6000 ---p 00002000 00:01 5490       /lib/libdl-2.24.so
1970-01-01T00:07:02.242+0000 F -        [main] 76de6000-76de7000 r--p 00001000 00:01 5490       /lib/libdl-2.24.so
1970-01-01T00:07:02.242+0000 F -        [main] 76de7000-76de8000 rw-p 00002000 00:01 5490       /lib/libdl-2.24.so
1970-01-01T00:07:02.242+0000 F -        [main] 76de8000-76dee000 r-xp 00000000 00:01 5439       /lib/librt-2.24.so
1970-01-01T00:07:02.242+0000 F -        [main] 76dee000-76dfd000 ---p 00006000 00:01 5439       /lib/librt-2.24.so
1970-01-01T00:07:02.242+0000 F -        [main] 76dfd000-76dfe000 r--p 00005000 00:01 5439       /lib/librt-2.24.so
[... cut ...]
1970-01-01T00:07:02.243+0000 F -        [main] 76fb5000-76fbb000 rw-p 00045000 00:01 7573       /usr/lib/libssl.so.1.0.0
[... 33 lines cut ...]
  path: /data/logs/mongod.log

# network interfaces
net:
  port: 27017
  bindIp: 127.0.0.1,192.168.88.30

#processManagement:

#security:

#operationProfiling:

#replication:

#sharding:

## Enterprise-Only Options:

#auditLog:

#snmp:
```

I've tried other mongodb version (3.3.9, 3.3.15, 3.4.5) but none of them works. I'm currently building an older version.

msoula, 2017-07-14:
Just an update: I tried with mongoDB 3.2.15 too and still the same error and stack trace. I think the problem might concerned an rpi3 config, so I'm now going to try to build a new OS for a rpi1 config...

msoula, 2017-07-17:
Raspberry 1 config didn't work. I've made some try and error last week-end and everytime I got this segmentation fault result.

So, I checkout to 2016.02-rpi tag which is the last known version that worked for me. I updated original mongodb package to 3.3.4 configuration and started a new build. I finally made this thing work !

It has to be noted that I've to chroot my environment to debian jessie since build fails in debian stretch environment.

wiltonsantana, 2017-09-29:
Hi @msoula,

I've gotten this same issue in my application. I've found out two ways to solve this problem. First, I've downgraded the MongoDB version from 3.3.4 to 3.2.16, the last released version that officially supports arm 32 bits and followed this [article](http://www.clarenceho.net/2015/12/mongodb-307-segmentation-fault-on-arm.html) to build with _--opt=off_ I also had to add --mmapv1=on to build. It worked well but it involves disabling code optimization which is not interesting for my application. So I tried version 3.2.16 with --mmapv1=on and it worked too(I think they fixed this issue somewhere between the 2 minors). I'm testing with a minor release from Sept., but from initial analysis seems to be working.

gamaral, 2018-01-20:
Report to upstream (Buildroot).

closed 2018-01-20 by gamaral